nefarius/ScpToolkit #682: ScpToolkit Driver Installer stuck

therealdio, 2017-03-28:
[image]
It's been stuck like that for the past 30 minutes. What do i do?

therealdio, 2017-03-28:
[image]
I also get this when I connect my ds3.

Kazekai, 2017-04-07:
I have the exact same problem all the time, especially when I need to reinstall the damned thing because for SOME REASON, it constantly uninstalls itself or won't work at all when I plug in my controller. Restarting the PC is the only solution I know of. You can also try killing the service but that usually results in the service not being able to start again and requires a reboot anyway.

HyeVltg3, 2017-05-13:
Fixed this issue by Disabling Driver Signature requirement
Google how its done for W10
On W7, press F8 at boot (beep)
choose last option "Disable Driver Signature" (worded differently)
now SCPtool Driver Installer should finish.
Working perfectly for me now.
leaving this here as this page was one of my google hits.

closed 2018-01-26 by nefarius