nefarius/ScpToolkit #689: Systray crashes with new Windows 10 Creators update

mmbossoni, 2017-04-01:
Systray is crashing in Windows 10 Creators Update on startup

arr0ww, 2017-04-01:
I get the following errors when trying to install the drivers on Win10 Creators update:

Source: .Net Runtime
Application: ScpTrayApp.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: exception code c0000005, exception address 00007FFDFF7AFD0C

Source: Application Error
Faulting application name: ScpTrayApp.exe, version: 1.6.233.16006, time stamp: 0x568d9fdb
Faulting module name: d3d9.dll, version: 10.0.15063.0, time stamp: 0x631de416
Exception code: 0xc0000005
Fault offset: 0x000000000000fd0c
Faulting process id: 0x24ac
Faulting application start time: 0x01d2ab0579a79d0e
Faulting application path: C:\Program Files\Nefarius Software Solutions\ScpToolkit\ScpTrayApp.exe
Faulting module path: C:\WINDOWS\SYSTEM32\d3d9.dll
Report Id: 13f8a05f-f699-4a07-bd8b-cc267001ea9b
Faulting package full name: 
Faulting package-relative application ID:

Awebb, 2017-04-12:
Furthermore, since the Creators Update, the driver install utility closes itself silently a second after launching. The exe is visible in the task manager for the brief duration of that second. No error message is shown.

webdesignian, 2017-04-13:
Also having this issue in Windows 10 build 1703 (Creators Update)

deadair3210, 2017-04-13:
Not having any issue with ScpToolkit, install survived Creators Update perfectly fine. I don't use bluetooth so I don't know about that though

webdesignian, 2017-04-13:
I also don't use bluetooth, but having same issues as posted above.

mmbossoni, 2017-04-14:
Clean install of 1703: Driver installer does not work with error "unable to find libwdi"
Systray works fine though.
How I hate how Microsoft breaks everything from time to time.

Awebb, 2017-04-15:
1. A clean install with an older Windows image, then pulling the Creators Update via the windows update: The SCP driver installer does not work.
2. A clean install with the most recent Windows iso including the Creators Update: SCP driver installer works.

deadair3210, 2017-04-15:
Definitely smells like a bug in the upgrade process for the Creators Update is screwing something up that SCP doesn't like

arr0ww, 2017-04-15:
I can confirm that a fresh install of the Win 10 Creators update will allow you to install the SCP drivers. The drivers install without issue and both USB and Bluetooth are working.

ripps818, 2017-04-16:
I haven't gotten the update yet, but has anybody tried uninstalling and wiping SCP as well as going into device manager and manually removing any drivers associated with the bluetooth and controller. And then after rebooting, installing SCP again?

tombi999, 2017-04-17:
Fixed the problem by closing RivaTuner statistic server , now the drivers work again. There is also a new version of RTSS out that supposedly fixes problems with the windows 10 creator update.

mmbossoni, 2017-04-17:
OMD.
Closing Riva server fixed for me as well.
Thanks for finding the problem.

mmbossoni, 2017-04-17:
Just to keep the thread up to date, after updating RTSS for v7.0.0 everything is working fine.

webdesignian, 2017-04-18:
Confirmed, updating RTSS to the v7.0.0 beta allows SCP to run without issue.

Thanks for sharing, @tombi999 !!!

Awebb, 2017-04-20:
I can also confirm Riva to be the culprit. The old version prevents the driver installer from working, the 7.0.0 update works.

closed 2018-01-26 by nefarius