nefarius/ScpToolkit #714: Losing bluetooth driver everytime I shutdown my computer

g819072586, 2017-04-16:
I lose my PS3 gamepad bluetooth driver everytime I shutdown my computer.So I have to install the driver  evertime if i want to use my PS3 gamepad wirelessly.

Dtxtream, 2017-04-19:
I thought it was the Windows 10 Creators update that was causing this to happen.
Now that you mention it, whenever this happens, the SCP driver manager application launches on startup and starts using almost 70% of my CPU.

One thing to note is that I don't even use bluetooth when connecting my PS3 controller. I just use the wired connection instead, yet this tool still seems to install the Bluetooth driver anyway.

deadair3210, 2017-04-20:
Make sure that you are un checking the Bluetooth driver when you run the installer, either that or you can remove the Bluetooth hardware if its fesable to do so when you are installing the driver

g819072586, 2017-04-20:
@Dtxtream Above all, I am using the Bluetooth chipset integrated on my WIFI adapter.It is not an independent one.I heard that this problem does not seem to happen on an indenpendent bluetooth adapter.
 Actually,I am not sure that whether my blutooth driver is lost after shutting down my computer.What I can see is the bluetooth MAC address in  the Scp monitor turns in to "ERROR".
If I have no bluetooth driver,the address should be "Disconnected".
This problem also appear on Win7.I do not think it is caused by Win10 Creators Update.
I think this problem is related to Windows itself.
Do you know Motioninjoy? It is a software like SCP. It can also connect my PS3 gamepad to my computer via bluetooth. But the bluetooth address in it will be "00:00:00:00:00:00" after shutting down, whichis the same as "ERROR"
What's more , Why the PS3 gamepad can not be connected to my computer via bluetooth directly as a bluetooth device without driver like a Bluetooth mouse?

g819072586, 2017-04-20:
@deadair3210 
How can I use my PS3 gamepad wirelessly if I remove the bluetooth driver?

closed 2018-01-26 by nefarius