nefarius/ScpToolkit #715: 2 gamepad

visolncev, 2017-04-16:
I need help guys... 

I have 2 gamepad... DS3 and DS4. So I use DS4 my brother use DS3... Problem is when my brother switch on DS3 controller it's work on my pc. Cuz I have bluetooth dongle and useed DS3 controller before.  How can I uninstall DS3 controller? I tried evething What I can.

Please help

Fastback460, 2017-04-16:
If I'm reading your post right, the DS3 controller is automatically connecting to your pc even though you don't want it to? If that's what's happening then hold the reset button on the back of the controller for about 5 seconds. Or if you have a PS3 (which I'm assuming you do if you have a DS3 controller) then just plug the controller back into the PS3, then it should automatically connect to the PS3 until you plug it into your pc again.

visolncev, 2017-04-16:
I have no PS3. I will try make to reset

closed 2018-01-26 by nefarius