nefarius/ScpToolkit #719: windows 1703 problems.

TristisOris, 2017-04-17:
[image]
[image]
[image]

DarkMatterMatt, 2017-04-17:
I don't think the drivers will ever be signed. You can avoid this error by following the steps linked below.

https://www.drivereasy.com/knowledge/solved-windows-requires-a-digitally-signed-driver/

Simply Google `a digitally signed driver is required` and the first result is the solution.

TristisOris, 2017-04-18:
yep, it helps, partially.

Another trouble with SCP service registration - it doesn't start. I reinstall program in safe mode with disabled driver signature verification - it works.

But it doesn't work without safe mode. 2-3rd screenshots.

TristisOris, 2017-04-18:
disabled secure boot, usual windows start.

ScpToolkit Settings Manager begin work - before, not.
SCP service don't started automaticaly, but manualy i finaly run it.

[image]

Looks like problem is solved.
So, 1st trouble - secure boot. 2nd - driver signature.

closed 2018-01-26 by nefarius