nefarius/ScpToolkit #722: PS3 Controller not working after Win10 update (1703)

Magnarok95, 2017-04-18:
Controller stopped working after the new Windows update. Tried reinstall SCP and it looks like it can't find my connected PS3 controller in the driver installation window. Even though Windows itself recognizes it as I can find it connected in Windows own controller overview.

Magnarok95, 2017-04-18:
I got it to work now. Reinstalled SCP and tried to download the driver for the ps3 controller earlier. Then my controller would not show up in my DualShock 3 dropdown menu in the DriverInstaller. 

A couple of hours later my controller showed up in the dropdown menu. (I don't know why, didn't change anything.) And it worked after the driver installation.

Wolf0202, 2017-04-19:
Can you provide more information about how you were able to reinstall it?
I have the same issue and I already uninstalled the ScpToolkit and when I try to reinstall it even with Admin permissions, everything goes fine until I try to use the Installer that puts every file required and synchronizes the Bluetooth dongle, etc.

closed 2018-01-26 by nefarius