nefarius/ScpToolkit #724: Rolled Back Windows 10 Update, Now Cant Reinstall ScpToolkit

HyeVltg3, 2017-04-18:
Wish I took a capture of it, but When reinstalling from ScpToolkit_Setup.exe it states it cannot remove  the old install and to contact my technical support.

When I Uninstall from Programs and Features, it request an .MSI from a certain locaiton that doesnt exist anymore.
I just now tried Force uninstall using Wise Uninstaller and now Scptoolkit is gone from programs but the ScpToolkit_Setup.exe still says it cannot remove the old install.

KRiTZV, 2017-05-13:
I had the same problem but I didn't force uninstall it, solved it by downloading and running the setup for the version I installed with, if its the right version it'll give the option to repair, modify or uninstall.

Niamrej, 2017-05-29:
I have the same issue, except ScpDriver.exe won't even attempt to run. I can briefly see it in TaskManager but that's it.

closed 2018-01-26 by nefarius