nefarius/ScpToolkit #725: When using SCP drivers Bluetooth will not work anywhere aside from my PS3 controller.

NickyNice, 2017-04-18:
I've been trying to connect my phone to my PC through bluetooth so I can use the creator update new "Dynamic Lock" feature. When I try and turn on bluetooth it is not available. If I run the troubleshooter it tells me my PC doesn't have bluetooth and to go buy a dongle.
  
So if I uninstall SCP drivers and just have the normal bluetooth driver it works, but can't connect my PS3 controller to it. Once I install the SCP drivers bluetooth stops working normally and will only work with my PS3 controller. I've tried uninstalling/reinstalling the drivers several times. Any other ideas?

growmaster112, 2017-04-19:
When you install the Bluetooth driver you are replacing the current/default drivers, which means the bluetooth dongle doesn't work with normal bluetooth devices.

n0xew, 2017-04-19:
This is the normal behaviour as pointed out by @growmaster112

NickyNice, 2017-04-19:
Ah well thanks for filling me in!

milothefirst, 2017-05-09:
I have this exact problem, except I am not even trying to use the bluetooth feature.

when I installed the driver for my DS3, my (inside built) bluetooth device disapeared from the device manager, like wise I can't access it elsewere, thus I concluded my pc has stopped recognizing it as such. the thing is I have unchecked all boxes on the bluetooth driver section of the SCPtoolkit driver installer (also including the main "install bluetooth driver" box of course). I can revert this by using the former program to uninstall the bluetooth driver but this also unistall the driver for the controller (which I have only been using through usb cable anyways). 

Esentially, my problem seems to be, even when I only select to install the plugged-controler driver the bluetooth driver also gets installed, likewise when i choose to uninstall the bluetooth driver  the controller's driver also get uninstalled. Is there anyway to fix/prevent this? it is really anoying to manually uninstall the bluetooth driver when I am done gaming (I have to use my bluetooth for other stuff) then reinstall the driver next time, rinse and repeat. since I am not particularly interested in using my controller wirelessly giving up on the feature entirely is completly aceptable.

Thanks for looking into this

batmanlovesuperman, 2017-06-05:
@milothefirst same here

rilwal, 2017-06-22:
I'm having the same issue, the Bluetooth driver is installed regardless of what devices are selected on the installer.

ice2hot, 2017-08-27:
Has anyone tried using two different bluetooth devices such a bluetooth 2.0 and bluetooh 4.0 For Scp drivers I believe you need a bluetooth 4.0. So would it be possible to install two bluetooth drivers and then using bluetooth stack switcher to switch between blue tooth drivers. 
Please share your views. Thanks

much-wow, 2017-10-20:
So if I want to use bluetooth I just uninstall scptoolkit or will I have to do something else?

closed 2018-01-26 by nefarius