nefarius/ScpToolkit #726: (Windows 10 Creators Update 1703, Dualshock 3) Tray Notification Crashes and Driver Installer does not work

DoctorCandor, 2017-04-18:
Upgraded my desktop to 1703 - my Dualshock 3 is no longer detected, trying to restart the Tray Notification crashes it, nothing appears after running the driver installer. Reinstalling changes nothing. Rolled back to 1607 and works just fine after a repair installation - I'll stay there until there's a way to work around it.

novedevo, 2017-04-19:
From what I can tell, it is because SCPToolkit uses .NET.  Microsoft updated to .NET 4.7 in the creators update and this broke compatibility with SCPToolkit.  Attached are two snips from Windows Event Viewer, caused by the tray application crashing when it tried to do anything.
[image]
[image]

DoctorCandor, 2017-04-19:
Huh. Hope that's the only incompatibility going on here.

novedevo, 2017-04-19:
There also might be something going on with directX.  One of the errors there mentioned d3d9.dll.  I reinstalled dx9, but no change.

DoctorCandor, 2017-04-19:
I see. I did try the more recently updated solution with FireShock + ViGEm on 1703, but couldn't get it to work and the documentation wasn't very clear on how to properly install everything. I managed to blue screen after trying to reinstall drivers included with ViGEm and just decided to roll back to kill two birds with one stone. Best I can do for now is to wait. Heard around on the web that 1703 is causing more problems than 1607 did.

closed 2018-01-26 by nefarius