nefarius/ScpToolkit #727: memory leak (Windows 10)

dogtorque, 2017-04-20:
I noticed that rarely scptoolkit will have a memory leak. and the only way to stop the memory leak is by shuting down the program through task manager. this has only happened twice now in the span of 2-3 months and I don't know exactly what causes it but I thought I would let the devs know.
I'm using scp driver installer 1.6.238.16010 and I'm using windows 10 (10.0.14393.0)

dogtorque, 2017-04-24:
I just found out that if you put the computer into sleep mode enough then the program will eventualy have a memory leak.

prkiso, 2017-10-20:
I have a similar issue - posting here in the belief that the two are interconnected.
I have noticed that when I leave my computer running for quite a while (~24 hours), SCPToolkit ends up consuming 4GB of memory. The CPU usage is similar for some reason also.

closed 2018-01-26 by nefarius