nefarius/ScpToolkit #736: Installing PS3 driver keeps killing my mouse after W10 Creator Update

Arcterion, 2017-04-26:
So after I got the Creator Update I noticed that my PS3 controller stopped working every time I rebooted my computer. At first uninstalling and reinstalling the driver worked, but after a couple of times it started giving me a Bus Error. 

After one more reinstall it suddenly killed my mouse. After freaking out and pinpointing the problem, I went through the manual driver removal that was posted [here](https://github.com/nefarius/ScpToolkit/wiki/Manual-driver-removal) and managed to get my mouse back.

However, now every time I reinstall the driver, it keeps killing my mouse and on top of that my PS3 controller still doesn't work. :(

I haven't installed the Bluetooth driver seeing as I've plugged in my controller to my USB, so a lot of previous mentions of this problem don't seem to apply to my problem.

Arcterion, 2017-05-07:
Anyone? :(

Arcterion, 2017-05-11:
Uninstalling ScpToolkit completely for now. Bit annoying  I can't play half my games now though.

sheldon25, 2017-05-13:
Is your mouse bluetooth, regular wireless or wired? If it is bluetooth, the driver could be overwriting that bluetooth dongle for controller support.

Arcterion, 2017-05-14:
My mouse is wired, using USB.

sk4rlight, 2017-05-21:
I'm having a similar problem, and I work around it by unplugging the other USD devices (mouse, keyboard) before trying to install SCPToolkit. Then it didn't seem to overwrite those drivers, and the USB devices kept working normally.

Arcterion, 2017-05-23:
@sk4rlight If I unplug my mouse and keyboard I have no way to start the installation though. :(

sk4rlight, 2017-05-23:
A cheeky suggestion that is not guaranteed to work but worth a try imo:
1. before u launch the Driver Installer try unplugging both the mouse and KB, then launch it (so their drivers don't appear in the list? not sure if it makes difference though)
2. then plug in your mouse and select the controller drivers u want to install, then highlight the "Install" button, so you can just press Enter on KB to initiate it.
3. unplug your mouse, leave KB on, then just press Enter
4. if it fails you could probably still use your mouse to revert it

Arcterion, 2017-05-24:
@sk4rlight Yeah, but like I said, if I unplug both my mouse and my keyboard I will have no way to actually start the program...

closed 2018-01-26 by nefarius