nefarius/ScpToolkit #740: Problem with the bluetooth feature

NotSnowy, 2017-04-29:
Hi! Welcome to my history.
To be clear i'm using a fake ds3 controller.

I've been using SCP Toolkit for a while with my fake DS3 (I'll call it FDS3) and a cable (the cable still works fine), sometime later I thought "well is time to go to the next level" 
I bought a doodle for my desktop pc to connect the FDS3 with bluetooth. I went to the program, press Bluetooth instalation, I select the bluetooth device (I'm sure that I didn't make a mistake selecting the device), clicked the force instalation option, and install.
So, here is the problem: With the FSD3 without cable, I press the PS button aaaaaaand nothing.
I opened the SCP Monitor and I saw that when the FDS3 is off the SCP Monitor says "Pad One : DS3 00:26:5C:CA:DD:00 - Reserved" but when I press the PS button it says "Pad One : DS3 00:26:5C:CA:DD:00 - Bluetooth 00000000 None" and when the FDS3 get tired of led blinking and turn off by itself, the SCP Monitor goes back to "Pad One : DS3 00:26:5C:CA:DD:00 - Reserved"

So, after that I saw so many videos to see what I'm doing wrong, and I couldn't find anything, the only lead I have is that the problem is by my fake DS3

I'll be so glad if anyone can help me with this problem, sry for the english, SCP team I love ur work and greetings from Argentina!

veeectorm, 2017-05-18:
Having the same problem down here buddy.  It seems that the problem might be something else, as i have both fake and genuine controllers, and im experiencing the same exact issue.
It has to be something else...bluetooth wont work at all.

W10, 64b here.

NotSnowy, 2017-05-21:
Sry for the late reply
Same, I have W10 64b, y looked for the compatibles bluetooth devices (https://github.com/nefarius/ScpToolkit/wiki/Compatible-Bluetooth-Devices) and found that my doondle is compatible (USB\VID_0A12&PID_0001) but also seems that the doondle has to be compatible with EDR (Enhanced Data Rate) and I don't know if my doondle has that.
I'll try buying a new one

closed 2018-01-26 by nefarius