nefarius/ScpToolkit #751: When installing the controller's driver the bluetooth driver also gets installed even if uncheck its boxes

milothefirst, 2017-05-09:
when I installed the driver for my DS3, my (inside built) bluetooth device disapeared from the device manager, like wise I can't access it elsewere, thus I concluded my pc has stopped recognizing it as such. the thing is I have unchecked all boxes on the bluetooth driver section of the SCPtoolkit driver installer (also including the main "install bluetooth driver" box of course). I can revert this by using the formerly mentioned program to uninstall the bluetooth driver but this also unistall the driver for the controller (which I have only been using through usb cable anyways). 

Esentially, my problem seems to be, even when I only select to install the plugged-controler driver the bluetooth driver also gets installed, likewise when i choose to uninstall the bluetooth driver  the controller's driver also get uninstalled. Is there anyway to fix/prevent this? it is really anoying to manually uninstall the bluetooth driver when I am done gaming (I have to use my bluetooth for other stuff) then reinstall the driver next time, rinse and repeat. since I am not particularly interested in using my controller wirelessly giving up on the feature entirely is completly aceptable.

Thanks for looking into this

dashmoopies, 2017-05-09:
Sorry to start asking you a question even though your problem isn't solved, but I have the same problem as you where the Bluetooth just disappears from Windows and it is nowhere to be found. How did you get the Bluetooth to reappear into Device Manager and be usable again?

milothefirst, 2017-05-09:
Hi dash, there is no need to apologies. it is my pleasure to be able to contribute to the community with what little knowledge I got

First open the SCPtoolkit_driver_installer (open it as admin just in case), then uncheck all install [ ...  ] driver boxes except "install bluetooth drive", click the "choose bluetooth dongle to install" and in the list check only the box that has the name of your bluetooth device next to it (mine says "RT bluetooth radio) then click uninstall.

that should do it, in order to check if it worked go to device manager, you should be able to see the bluetooth sub-menu again, if not try ordering to look for hardware changes.

luyong20121221, 2017-06-11:
i was trying to use "wii remote controller” to play in Dolphin (wii emulator) some day before,
I have a DS3, a wii remote controller. And of course , only one CSR 4.0 bluetooth adapter.
here is what I did:
1.	install scptoolkit normally, now I can use my DS3 as the native xbox 360 controller in windows 10.
2.	Open Device Manager in windows 10 , uninstall “generic Bluetooth radio” (this is exactly what the CSR Bluetooth adapter is after you install the scptoolkit) with “Delete the driver software for this device” checked. 
3.	Restart your PC. From now on ,the DS3 can only be used via usb cable ,and the Bluetooth ICON comes back to your taskbar.

closed 2018-01-26 by nefarius