nefarius/ScpToolkit #753: Forza Horizon 3 No Longer Detecting DS3 Controller

boskio, 2017-05-10:
When I first found this in October 2016 it worked a beautiful charm.. Unfortunately the game was awful but I'm trying to go back to it today, only to find my controller is no longer being recognized and I don't even know what anyone's talking about when they say "check SCP Server" etc because I've not a clue where to find anything like that. The drivers are installed fine but I can't even find out where I see if it's on. All I can do is check it in Driver Manager and it says both drivers are active, so really there shouldnt be a reason for it not to be working. Any ideas?..

boskio, 2017-05-10:
Update: Found the server, get the error "FATAL: The root hub is already running, please stop the scpservice"...

jacobmix, 2017-05-10:
Try typing this in a (admin) command window?
`net stop "SCP DSx Service"`
Press enter, and wait till it's done.
Then type this:
`net start "SCP DSx Service"`
Press enter, and wait till it's done.
That should restart the service.
Hope that helps.

boskio, 2017-05-10:
I've ran into nothing but problems, been trying all day to sort it. I used the cleaning service and wiped it, also wiped all drivers I could find as far as I'm aware, my controller does not show in Game Controllers tab but in Device manager says it's working perfectly, and in the SCP Profiler, works flawlessly. I do not understand why it isn't working?.... It's really grinding my gears now.

boskio, 2017-05-10:
Also tried to troubleshoot the device in Printers & Devices, It tells me the device might is older and might not work with USB 3.0, and did not fix, tried it in a 2.0 (i think) and still nothing. I don't see why it would matter because It worked fine in my USB 3.0 in my front panel when I first got it. I loved this driver because of how simple and easy it was, I must've been seeing something else! What's going on!? 

I don't know if its SCP or me! I somehow got it to show up in game controllers for literally a minute before it disappeared. I uninstalled SCP then wiped it, reinstalled the normal driver by plugging it in and it was in there.. I reinstalled SCP and it disappeared.. so frustrating!!

jacobmix, 2017-05-10:
Does it work in other games? Also some general PC specs would be helpful. Like OS, ect.

boskio, 2017-05-10:
Ah apologies, ofcourse.  I've not actually tried any other games, I don't really know if I have any, I only ever used it for Forza back in the day. 

ScpToolkit v1.6.238.16010 (ALSO tried some beta version I found but I think it was older) 
Wired DS3 Controller (Official Sony PS3 Controller) 
Windows 10 Pro Version 1607

PC Specs ( If Needed I don't know ) 
CPU: i5 6600k 
GPU: EVGA 970 SC GB ACX2.0
MB: MSI Z170A TOMAHAWK 
RAM: Corsair Vengence LPX DDR4 3000mhz 16GB
PSU: Corsair CS750M

jacobmix, 2017-05-10:
Try updating to the latest version "[ScpToolkit v1.7.277.16103-BETA](https://github.com/nefarius/ScpToolkit/releases/tag/v1.7.277.16103-BETA)"
Click on "Releases" to see all.
Only thing that's not really fleshed out is the Profile Manager.
Tho sometimes the Scp Service doesn't start for me.
I have a shortcut to start it tho.
Think nefarius is done with ScpToolKit tho. No update for a long time.
He's also working on FireShock, and ViGEm. Which kinda replace ScpToolKit.
But they only work for Windows 10 right now.

boskio, 2017-05-10:
Ah yeah it was that one I downloaded I'm certain, I recognize "a stupid mistake". I really don't understand why it's not being picked up at all :/.. Sad ..

boskio, 2017-05-10:
Update: Doesn't work on Tom Clancy's The Division, I'm sure it probably should. It's still not in my game controllers so I think that's the issue. How I overcome that issue is a different story.. God I hate this.

boskio, 2017-05-10:
https://gyazo.com/ac4a1cc30a8cec7b7063a953a66f2283

https://gyazo.com/37e02896d88733f8d7bd984ca840fd69

Another update.. not sure If it's supposed to be there but under devices in "Devices and Printers" an Xbox 36 Controller for Windows is there.. not sure if this is an old driver or if it's there by default? 

https://gyazo.com/8dd6f5805178529027eca67467246aea

jacobmix, 2017-05-10:
So you've tried to delete all Hid, PlayStation, Xbox, ect. Drivers, and reinstall ScpToolKit with the latest version? (Right click in Devices and Printers, go to Device Manager, and check everything,
and right click, and delete anything that looks related.)
Also make sure you have all the Installation requirements before re-installing. Just to be safe. https://github.com/nefarius/ScpToolkit#installation-requirements

boskio, 2017-05-10:
I'll try for the 4th time, but yeah I done that. :( Ill get back to you in a second. Are you sure it's SCP toolkits fault this controller isn't showing up in Game Controllers? (especially the error about it not working?) I never had any of this before.

jacobmix, 2017-05-10:
Only thing i can think of. I don't expect your controller to be broken,
and the most likely fix is just to reinstall everything.

boskio, 2017-05-10:
Na neither are broken, I've tried two different cables and controllers. Neither work.

boskio, 2017-05-10:
By the way, when I initiate the uninstaller for the SCP, it still leaves all this nonsense behind. https://gyazo.com/a5e7661306d7a5c09b2e58de2739325d 
I deleted the folder last time but I doubt that worked, I know stuff goes into your appdata etc, so how do I completely remove it all?

jacobmix, 2017-05-10:
The best you can do it just run the uninstaller (Can be uninstalled from "Programs and Features")
Or the ScpCleanWipe.exe

boskio, 2017-05-10:
Yeah I done that before, I just reinstalled using the beta version and I have NO idea what happened but it works!! I seriously have 0 idea what I done, because I done nothing different to what I done earlier. Thank you so much Jacob for your help. Seriously appreciated mate :)

jacobmix, 2017-05-10:
No problem man ;) Remember to mark this issue as solved, and close it.

closed 2017-05-10 by boskio