nefarius/ScpToolkit #756: Reoccurring issues with DS3, SCPToolkit

sk4rlight, 2017-05-14:
Hello SCPToolkit dev team,

When I first discovered this SCPToolkit method to use my DS3 on PC, everything was working as intended for a while (the start was probably just a little bit more than 1 year prior to this post). Meaning that even after PC restart, whenever I plug my DS3 in it gets recognized and is ready to use and play games with.

Recently, for probably more than quite a few months now, everytime I want to use my DS3 controller with SCPToolkit I have to reinstall the drivers. Whenever I do this, I might even need to run it a couple of times to work properly (not to mention it sometimes disables my USB keyboard and wireless mouse in the process - in that case I just unplug them before I run the installation).

**So the issues summed up are:**

**1.** After every restart, my DS3 controller does not get recognized, and is not ready to use (all 4 red lights blinking)
**2.** There are times, when shortly after reinstalling the drivers (maybe within 5 or 10 minutes), the driver shuts down or some sort, making my controller unable to be used again, making me re-do the installation again until it works. Whenever this happens, usually a small error window appears with a red cross, stating something wrong with .NET Frameworks.
**3.** Even with every other USBs unplugged, there are times, when there are some other possible DS3 controllers detected, putting the real DS3 controller on the secondary position (I know I can move it up to the first using the Monitor program).

From the top of my head, these are the issues the come to my find.

Thank you for your great work, I appreciate what you guys are doing, and thank you for helping me!

[ScpToolkit_Log-Package_pxphu_1494780029.zip](https://github.com/nefarius/ScpToolkit/files/999566/ScpToolkit_Log-Package_pxphu_1494780029.zip)

jacobmix, 2017-05-14:
Have you checked if it's the service that's the problem for 1, and 2?
How to make a [shortcut](https://docs.google.com/document/d/116LwfIM6evk4Dy89N1EQjGYVbcgdIX3cHE8CIR1pflk/edit?usp=sharing) for the service.
For 3 have to checked if you got any "phantom/ghost" devices?
To uninstall open Device Manager, and make sure hidden devices are [shown](https://support.microsoft.com/en-us/help/315539/device-manager-does-not-display-devices-that-are-not-connected).
Then find your "phantom/ghost" drivers, and uninstall them.

sk4rlight, 2017-05-15:
Thank you very much jacobmix for the fast response.
It seems that your solution for my 1. and 2. problem by making a shortcut to manually start the service fixed them, I am very grateful for that!

As for 3: how do I locate what that phantom/ghost driver these are? All I know from the SCPMonitor is the address of that device or some sort: [http://i.imgur.com/BwONXAE.png](http://i.imgur.com/BwONXAE.png )(pad 2 should be the ghost, pad 1 is the actual DS3).
I have checked in Device Manager but I can't seem to find this information anywhere.

Thank you again!

jacobmix, 2017-05-15:
Does your Device Manager show all hidden devices? Should look like this:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
Connect all your controllers if you can, and delete the grayed out DS3 controllers.

sk4rlight, 2017-05-15:
There are occasions when I start the service manually it disables my USB keyboard and my controller still does not work. This is not consistent though, sometimes it happens, sometimes it does not.

I have deleted 2 greyed out "ghost" devices: 1 was a 3rd Xbox 360 Controller for Windows, and the other one was a greyed out Playstation DS3 device, both not on the list anymore:
[image]
The ghost device after uninstalling these still stays in the SCPMonitor.

As for the Devices and Printers section, all 3 of these controllers lead to this same advanced settings, showing that there is 2 controllers; the first one is not responding to the controller inputs, while the 2nd one does, and by default the 2nd one is chosen. This applies to all 3 of the devices in the Devices and Printers panel.
[image]

jacobmix, 2017-05-15:
Try disabling Xbox controllers 1 at a time in Device Manager till you find the one that's not working.
Then uninstall it. A restart might be needed to make sure everything's uninstalled right.

closed 2018-01-26 by nefarius