nefarius/ScpToolkit #758: installation result: WDI_ERROR_OTHER

benja1747, 2017-05-15:
hi, im trying to use my ps3 controller(original) on my pc but i get the above message during the driver installation, and also my controller is turned off, and you cant see it if you go check the scptoolkit monitor.my os is w10 pro 64bits, and under the devices and printers section i have confirmed that my controller is using some microsoft drivers and after running the scptoolkit driver those microsoft drivers remain unchanged, despite the fact that eve already disabled the enforcement of signed drivers, i also noticed that when i connect the controller 4 lights flash for a while and then they turn off and wont light up again. please help my fix this problem, ahead there will be some screenshots of the installation log and my ps3 drivers.
[image]
[image]
[image]
[image]

benja1747, 2017-05-19:
update: at the end i could´t find a solution to this problem, what i can tell though is that it had to do with drivers, the root of all this begun when my pc started recognizing the ps3 controller as an unspecified device and the drivers were some "libusbK" ones that apparently come with the toolkit but for some reason were working bad, i was able to update drivers from the ones my pc had and use microsoft ones, this made my pc recognize the controller again as such, and even games would recognize the controller was plugged in, but there was a problem lingering in the background that i could not identify that didnt allow the controller to send any inputs , rendering it useless, in the end i did a fresh install of windows 10 on my pc, formating the old partition in the process and set everything up from scratch and now it works just as it used to do before all this, hopefully the devs will look at this issue so nobody has to go to the same lenghts as i did to fix it.

Geowil, 2017-08-31:
Having the same problem with Vista 32-bit (fresh install).  Started vista in disabled driver-signing mode, everything loads correctly until it starts installing the driver.  It churns for a while and then spits out the above WDI_ERROR_OTHER error.

closed 2018-01-26 by nefarius