nefarius/ScpToolkit #759: ScpVBus: Driver signing no longer valid according to W10

JoshWobbles, 2017-05-15:
Driver certs need to be updated

Dienfive, 2017-05-16:
@nefarius please take a look!

Dienfive, 2017-05-18:
Hello @JoshWobbles @nefarius follow steps in this link, disable signature requirements, then no issue on signature!!!

https://www.wintips.org/how-to-fix-windows-cannot-verify-the-digital-signature-for-this-file-error-in-windows-8-7-vista/

sollano, 2017-05-22:
I share this same problem, and I really don't want to disable secure boot to install this. I wish there was another way to solve it!

nefarius, 2017-05-22:
@Dienfive No
@sollano there will be.

sollano, 2017-05-22:
@nefarius Thanks Man, glad to know. For the time being, if I restart with the special option to disable digital signature option, everything works fine with the latest release. So I'll just hang on until the next update.

cloud1250000, 2017-05-30:
is the update coming any time soon? I'm about to reinstall scptoolkit because it stopped working after the creator update.. should i wait for it? :D

Dienfive, 2017-06-10:
hello All, do you guys have any news?

JoshWobbles, 2017-06-12:
@cloud1250000 Your issue was unrelated, this issue is regarding expired signing.

Dienfive, 2017-06-19:
Hello All,

Still not had good news...

Dienfive, 2017-06-24:
has been waiting.... i can not disable secure booth pernamently, so everytime starting windows, i need to disable it and reinstall SCPtoolkit, nightmare!

WilliamFeely, 2017-07-20:
If you have an earlier licensed version of Windows install that then upgrade to 10.  This will put your OS in a "grandfathered" status authorized to load the driver.
As another mitigation if you can disable Secure Boot in the BIOS this will also ease the restrictions.  Just make sure if you use Bitlocker with a TPM chip that you pause protection before changing any BIOS settings or you will be forced to enter your recovery key.

closed 2018-01-26 by nefarius