nefarius/ScpToolkit #763: triggers l2 and r2  get mapped to z axis instead of being mapped to triggers

MrRobot2211, 2017-05-20:
Hello scp toolkit works perfectly for me except that when i check the mapping for the triggers in the windows manager they are sent to z axis. I read about this before but it happens independently of being connected via bluetooth or not.

LucaSoranota, 2017-05-21:
trigger button ARE Z axis even if you are using the real x360 controller

MrRobot2211, 2017-05-21:
Hi shouldn't they be mapped to triggers in the xbox1 controller? because they are the only two buttons that seem not to work when i play games .

MrRobot2211, 2017-05-21:
I have tested the same controller using mayflash device for xbox control emulation and l2 and r2  work just fine, the buttons are properly recognized and activated. The issue occurs when using scptoolkit.

MrRobot2211, 2017-05-21:
I have tried deleting scp and installing from zero . know the problem exist only when connecting through bluetooth. The bluetooth is 2.1, any ideas?

LucaSoranota, 2017-05-22:
for bluetooth the recommended one is 4.0 altough it's sometime work at 2.0 with some issues but for Z axis are normal because Z axis is from x360 and for trigger button.. meanwhile analog are using X and Y axis

closed 2018-01-26 by nefarius