nefarius/ScpToolkit #765: Service not working correctly

marysiamzawka, 2017-05-22:
Windows 7 x64 here.

I've been using the original SCP before and everything worked fine.
Got my hands on this version, so did a clean wipe and everything. 
My pad is detected by the program, by something had gone wrong in the process apparently.
Looks like the SCP service failed to install, even though I did restart after installing, then reinstalled it a few times again, and even went as far as to do another clean wipe.
[image]

The controller itself is not being picked up by either Game Controllers applet or Xbox 360 Accesories thing.

jacobmix, 2017-05-22:
Not sure if it's the same problem i've been having. But SCP Service doesn't start (Or crashed?)sometimes  for me.
I have a [shortcut](https://docs.google.com/document/d/116LwfIM6evk4Dy89N1EQjGYVbcgdIX3cHE8CIR1pflk/edit?usp=sharing) for starting the service tho, and works fine after i run it.

marysiamzawka, 2017-05-23:
> I have a shortcut for starting the service tho, and works fine after i run it.

Tried it, but the service just refuses to start.

Also tried to start it through Services applet, with same results.

[image]

(it says "Starting dependance service or group failed")

jacobmix, 2017-05-31:
Try going into services.msc to see if your SCP Service is called "SCP DSx Service" Might be different.
Also updated some info on the shortcut about this. You could also try "sc start" instead of "net start"
https://superuser.com/questions/315166/net-start-service-and-sc-start-what-is-the-difference
Not that i think it'd do anything. Try looking in services.msc or uninstall, reinstall everything again.
Hope you get it working. Sorry for late reply.

closed 2018-01-26 by nefarius