nefarius/ScpToolkit #767: How do you... start the service after you reboot?

Markeron, 2017-05-24:
Tray notifications is sitting there, isn't that supposed to start the service? do I have to install the driver every time I want to start up?

nefarius, 2017-05-25:
No and no, it's supposed to auto-start. That's why it's a service...

closed 2017-05-25 by nefarius

jacobmix, 2017-05-25:
Even tho i've made sure that the service "services.msc SCP DSx" is to automatically start.
It doesn't seem to sometimes. Or maybe just crashes?
[image]
I have a [shortcut](https://docs.google.com/document/d/116LwfIM6evk4Dy89N1EQjGYVbcgdIX3cHE8CIR1pflk/edit?usp=sharing) to start it easy, and quickly manually tho.
Hope this helps you start the service easier after boot.