nefarius/ScpToolkit #774: Indeca (fake) PS3 controller not working, vibrating all the time

Miridinia, 2017-05-29:
[2In1 USB Joystick_hid-report.dump.txt](https://github.com/nefarius/ScpToolkit/files/1035744/2In1.USB.Joystick_hid-report.dump.txt)

Well, there's the dump. When I use the current stable version, it installs the controller fine, but then it just doesn't stop vibrating until I plug it off. And then if I plug it on again, it keeps vibrating. However, it appears on Devices as an actual 360 controller, so it actually worked somehow? It doesn't react to any input though. 

When I use the beta version, it doesn't appear on the selection screen though, and scptoolkit is the only program where it even reacted at all. According to x360ce it just didn't exist, and I've tried many different releases of both versions and everything. 

I've seen there are some similar issues on this page, most notably #318 , but I couldn't understand what they did either to solve it. As you all probably know, there are no more official PS3 controllers being made, so buying an official one is kind of impossible. 

What should I do? I just want to play Tales of Symphonia :(

jonnralge, 2017-06-29:
I have the same issue and same controller I just want to play Dark Souls 3 :^(

benzarr410, 2017-12-27:
Same controller, same issue. I just wanna play Stick Fight: The Game :frowning:

closed 2018-01-26 by nefarius