nefarius/ScpToolkit #781: This stuff made all my other joypads stop working

robertollll, 2017-06-04:
after i installed this drivers which doesn t work on my window7 32 bit also my xbox 360 pad and my Qanba joystick stopped to function!! No way i can restore them even if i uninstalled the scp toolkit! Reinstalled xbox360 drivers without success! Where are the old versions? After Nefarius put his hands on it it becames a total mess

robertollll, 2017-06-05:
Please somebody responde, my qanba is definiltely dead

luyong20121221, 2017-06-11:
recommend you try Motioninjoy in windows 7 to use the DS3/4 as a xbox 360 controller.
And for your current issues, I suggest you uninstall all related drivers with “Delete the driver software for this device” checked. 
then retry.

closed 2018-01-26 by nefarius