nefarius/ScpToolkit #783: [Windows 10 Pro (Version 1703/Build 15063.296] ScpToolkit suddenly not working.

coachl4d2, 2017-06-06:
I am unable to open any of the apps from ScpToolkit, when I double click it does nothing. When I try to open Tray Notifications it just crashes every time. This is after the creator update for Windows 10 Pro [Version 1703 - Build 15063.296].

VitoMolas, 2017-06-06:
same thing happen to me after the update, just wonna say "Great job Microsoft".

ArtuHxD, 2017-06-07:
Same here, creators update fucked it in the ass with a broom. Hard.

batmanlovesuperman, 2017-06-07:
Wow i am too late

ArtuHxD, 2017-06-07:
And no, there is not anything you can do. (EXCEPT ROLL BACK AND LOSE THE FEATURES)

VitoMolas, 2017-06-08:
guys, I found out how to fix this. I have been using MSI afterburner and enabled the on-screen display, this is what's been making it crash, so disable it.

phamminh1998, 2017-06-08:
anyone try the solution @VitoMolas used, i have not been using MSI afterburner or any alternate but SCP still crash after Creator update

VitoMolas, 2017-06-08:
@phamminh1998 the solution I got is from Deamon Tools support https://www.facebook.com/notes/daemon-tools-software-community/daemon-tools-creators-update/10155019526560049 , apparently the update changed how WPF runs, you might want to look at anything that would modify/add to the interface application.

luyong20121221, 2017-06-11:
ok for me.
[image]

malwilson, 2017-06-22:
Any update on when this will be fixed? its pretty anoying to have to uninstall and reinstall scp driver every time turn on my computer.

iVarco, 2017-07-09:
Is there alternative Software which runs on windows 10 until this problem is fixed?

closed 2018-01-26 by nefarius