nefarius/ScpToolkit #784: Steam games detected PS3 controller even it is not plugged in

TrollingNewbie, 2017-06-07:
So instead of showing me keyboard/mouse, It actually shows controller guide (which button to use in game) ... 
It is so annoying tho, sometimes I don't want to use a controller to play game, and I found no way to turn off the ps3 controller detection

TrollingNewbie, 2017-06-07:
Forgot to mention I am running on a laptop, windows 8.1 64 bit

ArtuHxD, 2017-06-07:
The new steam update ruined most games. Reinstall drivers. Pro tip: DO NOT INSTALL CREATORS UPDATE! IT WILL RUIN SCP!

TrollingNewbie, 2017-06-07:
Thanks. Tho I found one thing, at least that works for me. Open the task manger and there is a xbox360 controller at the bottom of the list, disable all 4 of them and restart. It works : P 
And when you want to use controller just enable it. 
But it is still a bit annoying tho because you have to restart everytime if you want to turn off the controller.  But hey, I guess at least there is a solution for this **** mess.

Sonickyle27, 2017-06-09:
Tick the "Always unplug virtual controller" setting in the ScpToolkit Settings Manager. It's under the service settings tab.

closed 2018-01-26 by nefarius