nefarius/ScpToolkit #788: Works perfectly for one controller, not for all 4.

LucidFir, 2017-06-11:
First; thanks for making this amazing piece of software. I've got myself 4 @play PS3 controllers as they're on a very good deal at the moment. I have until Saturday to return them for a refund if I can't get them to work.

I got one controller to work in Steam, PCSX2, Dolphin, and Castle Crashers (using XboxCE). However I cannot get all 4 to work at the same time. I've completely uninstalled all the drivers and the installation software and started from scratch a few times, I've made sure the controllers are all plugged in for the duration.

Not sure what else I can do, I hope you can help.

Sonickyle27, 2017-06-11:
How are you trying to connect them? Bluetooth or cable?

LucidFir, 2017-06-11:
They are all cable.

It seems that they have some form of conflict. Either any controller after
number 1 doesn't connect or if it does it goes weird.

closed 2018-01-26 by nefarius