nefarius/ScpToolkit #794: Very long error message

ghoul10p, 2017-06-13:
I was using SCPToolKit for a few days with no issues. Then I decided to try DS4Windows but had issues with it when using multiple controllers, so I went back to SCPToolKit. Today I got this error and my PS4 controllers no longer connect when plugged in. I need to reinstall the DS4 drivers every time I plug a controller in for then to be recognized. Any way to fix this? Help is greatly appreciated.

[image]

luyong20121221, 2017-06-13:
Open
“C:\Program Files\Nefarius Software Solutions\ScpToolkit\ ScpSettings.exe”
Goto “Service settings” and uncheck “Force DS4 driver installation”.

ghoul10p, 2017-06-13:
I did what you suggested, but the controller still doesn't work unless I reinstall the DS4 driver every time I plug it in. If i unplug it and plug it back in it is recognized as a DS4 controller but doesn't function until the drivers are reinstalled.

ghoul10p, 2017-06-13:
Maybe the messages will help:

These are the messages that come up when I plug a controller in.
[image]

These are the messages that come up when I install the driver.
[image]

And these are the messages that come up when I unplug the controller.
[image]

luyong20121221, 2017-06-13:
if i were you, i will uninstall all above and do this:
1.plug in DS4 via USB .
2.install scptoolkit normally, then run scpdriverinstaller and check all of them on the left.
that should work.

luyong20121221, 2017-06-13:
keep in mind dude you cant use DS3 and DS4 via bluetooth at the same time。

closed 2018-01-26 by nefarius