nefarius/ScpToolkit #803: DS4 problems

Alunage, 2017-06-22:
After disconnecting DS4 on wired and plugging it back in, it won't work but can be resolved by restarting scp service, would like a fix for this.

But the main problem is when the DS4 DOES work on wired, it randomly freezes/hanging for a solid second or two, which can lead you to make mistakes when playing whatever game.

hiddenhokage, 2017-06-22:
I think I have a similar problem, although I haven't experienced any freezing/hanging yet.

My issue is quite simple. If I boot with my controller plugged in, everything works fine. If I unplug it and try to plug it back in, it doesn't work anymore. The only way around this is to plug it back in and restart my computer entirely.

Alunage, 2017-06-27:
Did you try to restart the scp service instead of restarting computer?

hiddenhokage, 2017-06-27:
Yeah, I have a *.bat shortcut on my desktop to do just that. It works, but it's still kind of a pain that I have to do that every time I unplug my controller, even if by accident.

closed 2018-01-26 by nefarius