nefarius/ScpToolkit #811: Update Nag

Nopey, 2017-06-25:
Upon logging in, I am greeted with a dialog box telling me that SCPToolkit is up to date. 
I think that the auto-run update check should be told to not message when it doesn't find anything, probably using a commandline argument to tell it that it is the auto-run update, and not the user running an update themselves.

15wileyr, 2017-06-26:
Yup yup, it is very annoying. like a lot. Also! if you ever get around to adding the ds4 mouse feature like ds4 windows has you would be the shit!

15wileyr, 2017-06-29:
there is an Scpupdater.ini file      set the update frequency to 0

Nopey, 2017-06-29:
That's not really a solution, because I want to know when there is a new version, just not be pestered when there's not.

15wileyr, 2017-06-30:
I'm with ya, but id rather just manually check that deal with THAT STOOPID DIALOUNGE BOX. ITS THE END OF THE WORLD

Deide, 2017-11-17:
What I don't get is how this is running in the first place. I've disabled the "scpToolkit Tray Notifications" in the Start-up list. I've deleted any remaining start-up entries in the registry that wouldn't show in the Start-up List (at least all I could reasonably find). I've stopped the ds3 and updater services from auto running and then even flat-out disabled them.

What is causing this to run?

closed 2018-01-26 by nefarius