nefarius/ScpToolkit #817: Playing with Keyboard as Player 1 and DS3 as Player 2

loofychan, 2017-06-26:
As a few other people have mentioned, it doesn't appear to be possible to map my Playstation controller (DS3) as Player 2.  I'm trying to play Overcooked with Player 1 using the keyboard and Player 2 using the Playstation controller.  I only have one controller.  Is there any work around for this? Many thanks for your help!

erickrod2003, 2017-06-30:
@loofychan same issue here before my pc got factory reset i was using an older version that allowed me to use keyboard as p1 and control as p2 but now i cant find that version and i cant set my control to player 2 when i go to scp monitor its grey'd out so i cant make it p2 :(

erickrod2003, 2017-06-30:
@nefarius please help

Elibrone, 2017-07-26:
same issue here please help

closed 2018-01-26 by nefarius