nefarius/ScpToolkit #820: DualShock 3 input delay over Bluetooth

dnohales, 2017-06-28:
Hello. I'm experiencing some annoyances when using my DualShock 3 over Bluetooth, I'm having quite noticeable delays on the input of all buttons and axis, both on button releases and presses (it takes lot of time, sometimes around 0.5 seconds to respond to a button press or button release), and also the axis get stuck in a value for a noticeable time even when I moved the axis to different position.

This doesn't always happen, sometimes I have 20 minutes or so of not noticing any delay, or probably a perception while I'm not doing things like aiming with the stick.

This doesn't happen at all while connected on USB. Actually, looking at the SCP Monitor's Input Tester, I see the input is more responsive (the blue bars changes are smoother) compared with Bluetooth on those lapses when the noticeable input delay is not present.

Do you think changing the Bluetooth dongle would help?
Please let me know if I can provide some sort of debug information.

I've reproduced the issue on ScpToolkit v1.7.277.16103-BETA and v1.6.238.16010.
USB Bluetooth dongle: `USB\VID_1131&PID_1001&REV_0373` (brand MSI).
Windows 10 (build 15063)

Thanks!

evilbananas, 2017-07-21:
I'm experiencing the exact same thing and guessing this is a common problem.  The controller works fine like you say for the first 20 minutes or so, but then there is noticeable input delay.

I have tried 3 seperate supported bluetooth dongles and have the same result, so that can be ruled out.

dnohales, 2017-07-21:
Dang. Well, fortunately on USB works perfect, I'm fine with my 2 meter cable for now :)

evilbananas, 2017-07-22:
I think I have a solution!  It looks like it is related to the power settings Windows uses on the USB controller your bluetooth dongle is attached to.  

Scroll down to hostiphur's post in the following link and follow his instructions:
https://github.com/Jays2Kings/DS4Windows/issues/62

Hopefully this works for you.

dnohales, 2017-07-23:
Thanks @evilbananas! Gonna give it a try and let you know.

closed 2018-01-26 by nefarius