nefarius/ScpToolkit #822: Driver Installer Program Won't Run

tyty4646, 2017-06-30:
I reinstalled this program multiple times with all components selected. I also tried with only default components to install, but still no success. Before I reinstalled Windows 10 on this PC, all of the SCPToolkit programs worked fine... until now. The only issue is that the driver installer will not run, but gives me a spinning wheel for a few seconds, and it seems that it just stops trying and gives up. Thanks for the help <3.

Here is a log file from the debug tool:

[ScpToolkit_Log-Package_Tyler_1498831937.zip](https://github.com/nefarius/ScpToolkit/files/1115645/ScpToolkit_Log-Package_Tyler_1498831937.zip)

RudiBrox, 2017-07-04:
Same problem here. Installation worked fined before I reset windows, now it just opens and closes again. 
Had a problem with another program that had the same issue but I forgot which one it was and how I fixed it.

RudiBrox, 2017-07-04:
Ah I remember now, you need to close afterburner and Rivatuner or put the installer as a exeption in rivatuner so it does not hook into the program.

closed 2018-01-26 by nefarius