nefarius/ScpToolkit #826: Driver Installer not opening

GAalexandres, 2017-07-04:
Hello, I reinstalled my OS recently and now I installed "ScpToolkit", but now the "ScpToolkit Driver Installer" doesn't start, also the "ScpToolkit Clean Wipe Utility" doesn't start. The "Scp DSx Serice" is not running and if I manually run it says: "Windows could not start the SCP DSx Service on Local Computer. Error 1075: The dependency service does not exist or has been marked for deletion.". I can only start "ScpToolkit Monitor (legacy)" and "ScpToolkit Stand-alone Server (legacy)". Oh and I don't have any yellow triangles in the "Device Manager". What should I do? Thanks in advance.

hanyiabc, 2017-07-05:
I'm having the same issue!

AttK0, 2017-07-06:
Same issue here! Tried installing different versions but it didn't help.

iVarco, 2017-07-09:
I have the same Issue, without having the RivaTuner or MSI Afterburner installed..
Im wondering that everyone got that issue has also either of the software installed..

can any other software cause this?

I also installed Windows 10 (latest creators build) without upgrading...

AttK0, 2017-07-09:
You could try turning off any other overlay software that you may still have running. For example programs like Steam, GOG's Galaxy client and EA's Origin all have overlay funtionality which I've turned off myself.

closed 2018-01-26 by nefarius