nefarius/ScpToolkit #830: Wired DS3 appears in Monitor, Device Manager, but not in-game; no input

bunbijou, 2017-07-05:
[image]
[image]

As the title states, as of a week ago, neither my authentic nor my fake DS3 register input in any game. I used to use both of them at the same time with no issue, until they stopped registering any input at all. This doesn't correlate with to any Windows update or ScpToolkit patch, as far as I know.
I've been testing with Dolphin, which I've never had problems using scptoolkit with.

[image]
As you can see, my DS3 doesn't appear in the list of devices like it used to.

[image]
It also doesn't appear in the legacy game controller explorer.

Also, the lights on my DS3s turn on just like they used to, they just don't register input.
I have attempted to reset my controllers using the button on the back, to no avail.
I have cleanly installed SCPToolkit and reinstalled on multiple occasions with no success.
I have alternated USB ports, with no luck.

Strangely enough, I've successfully been able to use a DS4, using the same ports and cords as I had used for my DS3s, but i'd rather find a lasting solution.

I'll be happy to provide and further disambiguation, if neccisary.

Log Package:
[ScpToolkit_Log-Package_bunny_1499293673.zip](https://github.com/nefarius/ScpToolkit/files/1126229/ScpToolkit_Log-Package_bunny_1499293673.zip)

Johngaojiahao, 2017-07-06:
I am having the same problem too, hopefully they will fix this soon :(

jamie-moffatt, 2017-07-27:
I had the same problem, I tried installing an earlier, non-BETA version of SCP toolkit and was able to get it working that way (https://github.com/nefarius/ScpToolkit/releases)

closed 2018-01-26 by nefarius