nefarius/ScpToolkit #833: fake ps3 controller detected ? i don't understand

trilippo, 2017-07-08:
Hi, i looked for a solution in this forum and on the internet but i don't understand!
i have 3 ps3 controllers. one is original and other two are bought on web. they are all dualshock 3. their are wireless too. the original one works perfectly on my pc with scp toolkit installed. other two controllers won't work, nor wired nor by bluetooth (not avalaible on my pc).
leds flash in all 2 controllers and never stop. i tried to understand reading some topics on this problem ... but i am not able to resolve this problem .. can anyone help me please !????
thanks to all ...

Elibrone, 2017-07-26:
You should reset the gamepad.

DiamondGames204, 2017-07-30:
I have the same problem.

closed 2018-01-26 by nefarius