nefarius/ScpToolkit #852: Screen won't sleep with PS3 controller plugged in

vicyuste1, 2017-07-23:
After the new creators update my screen won't ever go into sleep mode while the PS3 controller is plugged in via USB.
Laptop clevo p650rs, Windows 10 home edition, if that helps.

rocksinboxes, 2017-07-28:
I have the same exact issue. I have a Surface Pro 3 I5 8GB Ram with Windows 10 Pro Creators Edition .

kidlazer, 2017-07-31:
I also had the same problem, however I am trying to put the entire pc to sleep and not just the monitor, but this hangup should affect both cases. Anyway, I was messing with all sorts of fixes for this and may have stumbled upon a solution. In the device manager, I disabled the Playstation 3 controller under universal serial bus devices, restarted, then re-enabled it. For some reason, this allowed my PC to sleep again automatically. I'm not entirely sure why this worked, but it's worth a shot for the rest of you struggling with this.

hdlineage, 2017-08-11:
This seems to be an issue for all game controllers, not just PS3 controller with SCPtoolkit. 
It happens when you have a game controller connected with Geforce Experience installed.

Look here for more info:
https://forums.geforce.com/default/topic/1003032/geforce-drivers/monitor-wont-sleep-windows-10-creators-update/18/

kidlazer, 2017-08-11:
That's very interesting. It seems to not be strictly limited to Nvidia drivers though. I have a pure AMD setup so that cannot be my issue. I will however say that it does indeed appear to be a game_controller/creators_update bug that needs ironed out by Microsoft. 

Also of interest is that my workaround (disable, restart, enable) has continued to work for me. I had one issue where it reverted to a non-sleeping state, but performing those steps again has made it functional for a number of days now. I think a Windows update on Tuesday may have been what caused my reversion, but I was not paying enough attention at the time.

closed 2018-01-26 by nefarius