nefarius/ScpToolkit #855: DS4 wont connect to a PS4 after using it on my PC

sisPedro, 2017-07-25:
Searched among issues and used google but cant find anyone with this problem. I used a friends DS4 on my pc and when he took it home his PS4 wouldnt recognize his controller. Brought it back to my house and we tried it on my machine. It works fine. Tried it in my PS4 and it wont connect. Anyone knows anything about this?

arthurkomatsu, 2017-08-04:
Try to connect your DS4 with a cable in the PS4

sisPedro, 2017-08-08:
I tried! tried pretty much everything(at least i could google). The only
strange thing i found that bugs me is that i cant set the controller to
connectivity mode(Share+PS Button). I've actually tried tons of things and
opened the controller to no avail. Thanks for answering anyway!

closed 2018-01-26 by nefarius