nefarius/ScpToolkit #856: SCP Driver is "unsigned or damaged"

TechNGamer, 2017-07-26:
So I'm trying to use SCP, however, Windows won't run the driver which makes the service not run. Windows is stating: Windows cannot verify the digital signature for the drivers required for this device. A recent hardware or software change might have installed a file that is signed incorrectly or damaged, or that might be malicious software from an unknown source. (Code 52)
This has just happened and I have no idea what to do since reinstalling it doesn't work, completely removing it then reinstalling it doesn't work and I personally don't know how to disable enforce driver signature. However I do note that the file is signed and Windows is detecting the file as being signed. I've looked at many suggestions and tried it to no avail.

FragileHearts, 2017-08-04:
this is expected on latest version. you can access disable enforce driver signature going to advanced boot. google is your best friend

TechNGamer, 2017-08-04:
@FragileHearts I fixed it by downgrading the version. When I tried that I forgot I made this at all. But I can tell you that the latest pre-release driver is having this issue since that is the one I'm talking about.

closed 2018-01-26 by nefarius