nefarius/ScpToolkit #866: Fake PS3 controller doesn't work AFTER installing SCP drivers

lul1234, 2017-08-02:
Basically it works when I plug it in and I can play games with it but layout is messed up. After installing scp drivers it stops working.

LoLlYdE, 2017-08-02:
Is the controller youre using in the [compatible controllers](https://github.com/nefarius/ScpToolkit#compatible-controllers) list?

lul1234, 2017-08-02:
No I guess. I don't even know how  controller is  called tbh , literally just bought cheapest wired one from Argos to use on my ps3 or laptop

LoLlYdE, 2017-08-02:
Then theres a good chance it wont work with scp drivers. You said it worked before, so try uninstalling the drivers, reboot your PC and see if it works again

lul1234, 2017-08-02:
I meant it that it does work without SCP drivers , but layout is messed up. for example in FIFA triangle become square(And the RS doesn't even work in game) And it doesn't work at all after installing SCP drivers
Is there any other program like SCP for controller on pc?

LoLlYdE, 2017-08-02:
I dont know about a program that does what youre looking for. Maybe you can rebind keys in the options somewhere? I cant help you further than that.

closed 2018-01-26 by nefarius