nefarius/ScpToolkit #870: no games are recognizing my controller

LoLlYdE, 2017-08-03:
I tried un-/reinstalling the drivers a couple of times now, with the clean wipe utility and the force install. Every time my controller gets recognized as a "xbox controller" in "devices and printer" I can open it and see that windows itself recognizes the inputs (see picture below)
[image]

however no game actually takes the controller inputs, it acts as if no controller were connected.

Games tested:

For Honor
Dark Souls III
ibb&obb
Super Hexagon

because it's not working for multiple games, I'm pretty sure its not a game specific issue.

Also when trying to use x360ce, the x360ce itself recognizes the controller aswell and maps the inputs correctly, but when launching the game it still wont recognize the controller. 

I've ran the Info Debug Collector according to its instructions and hope it helps.
let me know if more info is needed.
[ScpToolkit_Log-Package_LoLlYdE_1501785886.zip](https://github.com/nefarius/ScpToolkit/files/1198276/ScpToolkit_Log-Package_LoLlYdE_1501785886.zip)

LoLlYdE, 2017-08-04:
update: Steam Big Picture appears to recognize my controller. 
It works while using Big Picture, and I can launch Big Picture from anywhere by using the PS button

Hoshimon, 2017-08-04:
I'm also havnig the same problem, neither Dolphin, GuiltyGear, BlazBlue, or any games I have are detecting the PS3 controller, but it appears under Windows Devices, and Windows itself can see the buttons being pressed.

NLmitchieNL, 2018-01-19:
I also can't find any solution to this. My Dualshock 3 works fine. My Dualshock 4 portrays this issue as well. But when playing Dark Souls III with BP mode, my controller input gets all messed up.

closed 2018-01-26 by nefarius