nefarius/ScpToolkit #872: PS3 Controller doesnt connect via Bluetooth.

Sakare, 2017-08-04:
I installed ScpToolkit and everything workes fine. I can use my ps3 when it is connected via usb cable. But when i unpluge the cable, my Bluetooth won't work with my Controller. The ps3 Controller 4 red dots just Keep blinking. I also tryed the ps3 Controller reset  button. Bluetooth Shows as Bluetooth Host (ScpToolkit). Any Help ?

FragileHearts, 2017-08-04:
https://www.youtube.com/watch?v=RuxXUKN1vz8

try using this fix

Sakare, 2017-08-04:
Sadly doesnt help!

FragileHearts, 2017-08-04:
Have you tried adding your bluetooth dongle Hardware ID into ScpCintrol.ini? Try using that, I fixed mine using this method.

Firstly, wipe your current driver installation using ScpCleanWipe
Then, add your BT dongle hardware ID into the ScpControl.ini. You can find your BT Dongle Hardware ID in device manager, under the "Details" Tab. On the drop down list pick Hardware IDs. Right click on the first value you find and paste in into the ScpControl.ini

Proceed with the installation as usual. This time when you get to the step where you install the BT Dongle, you will see the dongle id on the top right selection box. select that, and select initialize. If the ID goes down to the bottom right selection box, it mean you can use your dongle with SCP. if not, bad luck then.

Sakare, 2017-08-04:
It is alrdy on the ini.

JackCobourn, 2017-08-12:
Along with running the ScpCleanWipe, try uninstalling the drivers for the Bluetooth adapter in Device Manager. This, along with a few restarts, got it working for me.

Zcehtro, 2017-08-15:
I too am having this issue.
According to many, it's likely the Windows 10 Creator's Update that broke compatibility.
We'ed need the author, Nefarious, help.
Sadly, he hasn't been active on this for a long time... :(

closed 2018-01-26 by nefarius