nefarius/ScpToolkit #875: High CPU usage : 1 core at 100%

LaPomme57, 2017-08-06:
Hi,
I just made a fresh install of Windows 10 x64 on my PC and SCP Toolkit is now constantly using ~15% of my CPU (i7 4770K @3.5Ghz) only when my controller is plugged with SCP service started.

Here is a screenshot with global CPU usage,
[image]
and here one with detailed core usage.
[image]

Here is also the debug information as requested on the official thread.
[ScpToolkit_Log-Package_Pierre_1502024314.zip](https://github.com/nefarius/ScpToolkit/files/1203091/ScpToolkit_Log-Package_Pierre_1502024314.zip)

I hope this issue can be fixed

Zcehtro, 2017-08-14:
I too am having this issue.

Sonickyle27, 2017-08-23:
What controller are you using? How is the controller connected?

Zcehtro, 2017-08-23:
Genuine DS3 controller, by means of bluetooth.

LaPomme57, 2017-08-24:
DS4 Controller, using a cable (also tried with other cables)

Sonickyle27, 2017-08-24:
If you rotate one of the sticks constantly and move the mouse, does the cursor stutter?

LaPomme57, 2017-08-25:
No it doesn't

closed 2018-01-26 by nefarius