nefarius/ScpToolkit #879: Driver installer doesn't show my motherboard bluetooth any more v1.7

djexstas, 2017-08-08:
Hello, I used older version of toolkit before reinstalling win, now I can't find what version it was. So, installer saw the mouse and Dell Wireless bluetooth adapters in older version, now (1.7) it doesn't. I've tried to add ID to bluetoothHost.inf, but it don't see it anyway (maybe I was doing something wrong). I tried to find how to solve it in "Issues", found some posts dated 2016 but there was no way except waiting newer release. Maybe there is a way I can add device by myself? I heard that in 1.7 was added some exceptions for devices to not list them in installer, because some people killed their other bluetooth devices, such as keyboards etc. 
P.S. Sorry for grammar, I'm from Russia

Dell Wireless 1707 Bluetooth 4.0 LE Device (Qualcomm Atheros Communications)
USB\VID_0CF3&PID_E005
USB\VID_0CF3&PID_E005&REV_0001

sheldon25, 2017-08-12:
I have a similar issue but with the program not recognizing my ps4 controller when using the latest 1.7 release.. Try using version v1.6.238.16010. That should work for you no problem.

djexstas, 2017-08-13:
I have to reinstall drivers each time I reboot system or sleep ,when I use 1.6 (tried some different versions), because it doesn't connect via bluetooth any more after it boots :(

djexstas, 2017-08-13:
added ID to scpcontrol.ini, bluetooth was found but when I click "initialize" it doesn't go in bottom list from top, it stays uninitialized

djexstas, 2017-08-13:
If I will buy usb bluetooth dongle, will I have problems with cheap devices, cost about $5? 
[image]

jactripper, 2017-08-24:
I met the same problem, but now solved.
You can try to added Bluetooth adapter ID into file "BluetoothHost.inf" under "\ScpToolkit\WinUSB"

djexstas, 2017-09-18:
Now it tries to install but fails( I always have problems with making something work, it's my life, I think it will be easier to buy USB BT dongle, thanks for trying to help me :D 
[image]

closed 2018-01-26 by nefarius