nefarius/ScpToolkit #883: New Windows Update Killed ScpToolKit

Xirone, 2017-08-13:
Okay, so, from the other posts here the new windows 10 update has messed things up for the toolkit to fully connect your PS3 controller. For me, when I connect my controller all 4 red lights blink, as if it were trying to connect to a playstation when the controller isnt synced. Except they flash slower. I went to windows 10 settings>update and security>recovery and there should be an option to roll back to a previous update. This is true, only my computer updated while I was on vacation "more than 10 days ago". Therefore, I can't do a rollback. Any possibilities or alternatives to connect my controller so I can game and stream? Thanks!

ShadixAced, 2017-08-16:
Creators Update breaks SCPToolkit functionnalities. Some people would say "Try manually" but I already have these SCP drivers installed and it doesn't work. Some people say try with Scarlet.Crush's version.

ShadyGuyJose, 2017-08-30:
I have the Creators Update and no functionality is broken for me, using the latest version from Nefarius. However, I did have the previous drivers from a really old version (1.2.2.175)

closed 2018-01-26 by nefarius