nefarius/ScpToolkit #887: PS3 controller connects, no notification, and doesn't work

LadyDeath1138, 2017-08-16:
My controller worked with Fallout 4 before, but for some reason, I can't get it to work again. (My boyfriend is the one who installed it and used it months before we got together, and even he doesn't know how to fix this.)
I tried uninstalling the drivers and reinstalling them, and restarting, but it still didn't work. The controller seems to connect, but the game will not register the input, and I can't even test the input. Is there something I'm doing wrong?
Please help.

ShadixAced, 2017-08-16:
You are on which version of Windows ? Because it seems that Creators Update (rs2/rs3) releases breaks SCPToolkit.

LadyDeath1138, 2017-08-16:
I'm using windows 10. I don't know what version we're using. Unless the updater changes the version automatically.

ShadixAced, 2017-08-16:
Well, please see the issue I've made at #888 and try if it works for you, since it did work for me.

LadyDeath1138, 2017-08-16:
I am still unable to get the controller to work. I don't know what the problem is. Perhaps I did something wrong?

ShadixAced, 2017-08-16:
Have you used the Bluetooth one or the USB one (on Install)

LadyDeath1138, 2017-08-16:
USB. I can't get the bluetooth to work at all.

ShadixAced, 2017-08-16:
Seems odd... Go to SCPDriver.exe and below uncheck Bluetooth Install, and try "Force Install"

LadyDeath1138, 2017-08-16:
I get an unexpected error when I do that.

ShadixAced, 2017-08-16:
Hmm... Uncheck Force Install.

LadyDeath1138, 2017-08-16:
More errors

ShadixAced, 2017-08-16:
What is it saying ? Maybe I could find what tells your log.

LadyDeath1138, 2017-08-16:
Couldn't request bluetooth host. A device attached to the system is not functioning. (This may be the issue)

Unexpected error, Reference not set to an instance of an object

Couldn't initialize sound engine, Could not find part of the path. (This issue has been going on since I discovered that it wasn't working.)

ShadixAced, 2017-08-16:
Okay, I see what can cause it.

Go to your file explorer, right click on Your PC and click on "Manage"

Go to your Peripherics, and search for "Bluetooth" or "Universal Serial Bus" and try to search for "PLAYSTATION 3 CONTROLLER". Uninstall this driver, and then retry/=.

LadyDeath1138, 2017-08-16:
Its working!

LadyDeath1138, 2017-08-16:
Thank you so much!

ShadixAced, 2017-08-16:
No problem, enjoy !

Alwayswithproblems, 2017-12-30:
hey ShadixAced, I have the same problems that LadyDeath1138 was having, except it's the PS4 controller. yesterday I installed the USB and Bluetooth drivers on the scp toolkit, and the controller worked on steam and on the Witcher 3: Wild Hunt. and today I attempted to use the bluetooth functionality, needless to say I failed to get it to work, I even installed Sixaxis tool kit, and still couldn't figure it out. I then uninstalled all the scp toolkit drivers and reinstalled them, restarted my system, and now the controller doesn't work anymore, I uninstalled/reinstalled all the drivers, went to the system manager like you told LadyDeath1138, didn't find a PS4 controller driver. the controller doesn't work at all now when I press the sony symbol while connected via usb. but the controller is functional when I unplug it. help me please.

ShadixAced, 2017-12-30:
@Alwayswithproblems BT or USB connected ?

Alwayswithproblems, 2017-12-30:
It was USB connected. Thank you for the response but I finally figured out a fix. I uninstalled the SCP Toolkit entirely, but still kept the setup.exe. I then unplugged and plugged the controller, afterwards installed the SCP toolkit from the setup.exe, and checked Bluetooth and PS4 Controller on the toolkit driver installers. After installing them the controller let up (the blue light that emits when the controller is on) and it worked on steam and the fore mentioned game. The only thing off about this solution is that the PC reads the controller as an Xbox 360 one, even in the Witcher it displays Xbox buttons on tutorials. But at least I can play the game with a controller now.

closed 2018-01-26 by nefarius