nefarius/ScpToolkit #888: FOR PERSONS WHO USES WINDOWS 10 CREATORS UPDATE - PLEASE SEE BEFORE POSTING

ShadixAced, 2017-08-16:
Hello everyone, 

I am also using Windows 10 Creators Preview, and it breaks SCPToolkit functionnalities.

But now, I am here to give you the solution !

Please use Scarlet.Crush's version : http://forums.pcsx2.net/Thread-XInput-Wrapper-for-DS3-and-Play-com-USB-Dual-DS2-Controller (you download the normal version + update)

Step by step instructions : 
1) Extract the normal version wherever you want
2) Extract the update over it
3) Launch SCPDriver.exe, and click on Install (if Bluetooth can be used into your PC, why not install this driver too, but I haven't tested)
4) When it's finished, launch SCPServer.exe
5) Click Pad 1 and ta-da ! Now you can use your DualShock 3 pad !

Zcehtro, 2017-08-16:
I troubleshot it as much as I could and documented almost everything here:

https://github.com/nefarius/ScpToolkit/issues/886

But couldn't make it work.

Surprisingly, your suggestion worked! I am now succesfully playing with my official DS3 gamepad again.
Sad to see Nefarius abandoning his project.

ShadixAced, 2017-08-16:
Nice to see that it worked for you, enjoy !

Zcehtro, 2017-08-16:
Yes. Both cable and bt worked.

Zcehtro, 2017-08-16:
However, the server didn't record the controllers connected (the controllers did connect to the computer, but the interface didn't refresh with the controller data connected as they do with the monitor). 

The monitor did record them.

ShadixAced, 2017-08-16:
That's odd, because that's the contrary that happened to me xD

Zcehtro, 2017-08-16:
No prob. Either way, the server is essential because it's what is listening to incoming gamepad connection requests.

RedDuckss, 2017-08-19:
Didn't work for me. Followed steps exactly, tried both ScpMonitor and ScpServer and neither fixed the issue. My PS3 controller is recognized by Scp when plugged in and the rumble tests work fine, even without using this method and doing the normal install. However no games recognize the controller.

alfredox123, 2017-08-20:
this is weird, i have win 10 creators update and i dont have much problems with ScpToolkit v1.7.277.16103-BETA. but i notice sometimes my controllers desyncs for a moment and i dont know if my bt dongle is dying or if it is because of creators update or maybe the game :S this happens more often on dragons dogma when the game loads anew zone (fast travel or when entering a village or castle)

alfredox123, 2017-08-20:
gonna try this one to see if it works better than the ScpToolkit v1.7.277.16103-BETA

alfredox123, 2017-08-20:
with v1.2.2.175 i dont have the desync problem, working 100%. better than v1.7.277.16103-BETA

ShadixAced, 2017-08-20:
Well, looks like you are the lucky one xD, even if I didn't had problems for a long time, but Redstone 3 (RS3_release) did break SCPToolkit fonctionnalities. Maybe you aren't in RS3.

alfredox123, 2017-08-20:
im on windows 10 v1703 i supose is red stone 2 right?

ShadixAced, 2017-08-22:
Yes it is.

Sonickyle27, 2017-08-23:
I'm using [this release](https://github.com/nefarius/ScpToolkit/releases/tag/v1.6.238.16010) on the creators update and I am have no issues whatsoever. I used to be on the Fast Ring and after every build update just reinstalling the drivers from the driver installer program got the controller and bluetooth dongle working again.

ShadixAced, 2017-08-23:
Creators Update said to be running fine as far as I know, but latest RS3 is messing some things (Pre-Fall Update). 

Still, will add as method 3. @Sonickyle27 Can you still tell me in which version of Windows you are ? It started "bugging" for me since ver. 16251 (RS3)

mbc07, 2017-09-01:
Well, here's my two cents: I'm currently running build 16273 from Windows 10 Insider Preview program and the latest beta (v1.7.277.16103) fails to install because the digital certificate used to sign the VBus driver expired. I'm not totally sure on why it works, but disabling Secure Boot on the BIOS/UEFI settings allows the VBus driver to be used even through the certificate expired. 

Another way that should allow the VBus driver load is enabling Test Mode (run `bcdedit -testsigning on` with administrative privileges), but keep in mind that Windows will show a watermark (which can be disabled with Universal Watermark Disabler app from WinAero) and that some PC games that uses EasyAntiCheat technologu (e.g. recent Ubisoft games) will refuse to launch while the PC is in Test Mode...

ShadixAced, 2017-09-01:
Wow haha. So you have the same build as mine.
I do not have that kind of option "secure boot" on my Phoenix BIOS.
Being in Test mode all the time isn't that good for the PC, since it is used for testing.

Sonickyle27, 2017-09-01:
I'll post my build number later as I'm at work right now, but I know I'm running the latest build that is available on the Release Preview insider ring.

alfredox123, 2017-09-11:
well i got the desync problem again but this wasnt for the version i have, its because this softwares doesnt like games that make your cpu reach 100% usage. to fix this i have to put scpservice on priority above normal on taskbar and now i dont have the desync problem :)

Formedras, 2017-10-28:
Yay! Sonickyle's solution (using Stable release 1.6.238.16010) works for me, including the PCSX2 lilypad-SCP pressure sensitivity
I'm running Windows 10 Pro stable version 1709 (Redstone 3) build 16299.19.
(Now if only I could get Steam to not recognize the controller: I want it for PCSX2 only. Maybe MAME. I've got two XB1 controllers, a DualShock4, a Mad Catz fightpad and fightstick, and a Steam Controller for everything else.)

nicholas-barcelos, 2017-12-22:
There is something wrong, i followed every step and still just get **host adress disconnected** and display pad status as charging. No errors or anything. I've been trying for hours.

1. wipe utility + uninstall + (install + update)*
2. rollback windows
3. try (1.) again

*As decribed on thread and here

ShadixAced, 2017-12-30:
@nicholas-barcelos Try Sonickyle's solution.

Sonickyle27, 2018-01-15:
Yeah I'm still running that release on my new laptop with Windows 10 Pro version 1709. Still working perfectly. Sometimes Final Fantasy 14 can't see my controller after resuming my laptop from sleep but otherwise it works without issue, both wirelessly and over wire.

closed 2018-01-26 by nefarius