nefarius/ScpToolkit #892: DS3/DS4 Not Working

VladVidya, 2017-08-20:
Prior to installing ScpToolkit, Windows Device Manager will pick up my DS4 as "Wireless Controller" and DS3 as "PLAYSTATION3 CONTROLLER", though I can't use either, naturally. 
After installing ScpToolkit, they're shown as exactly the same in Device Manager, and during the setup of ScpToolkit, only my DS3 will be detected when installing the driver. After initializing it and finishing installation (wired only, I don't use Bluetooth.), DS3 is still unusable, and Scp's monitor won't even pick up my DS4. All that happens after plugging in DS4 is the lightbar flashes yellow once and turns off afterwards. DS3 gets registered by the monitor, but will not work in PCSX2, Dolphin emulator, PPSSPP, or any games.
I have uninstalled both controllers and the software many, many times, and even used the wiper utility, to no avail. Same result every time.

For what it's worth though, I've also tried x360ce and had similar results.

ShadixAced, 2017-08-22:
Please see issue #888 for probable solutions. I've did that issue for a reason ^^

VladVidya, 2017-08-28:
I probably should have mentioned this as well, but I tried all solutions I could find, including that one, to no avail. I appreciate your effort though.

ShadixAced, 2017-08-28:
Try by uninstalling your actual drivers and by reinstalling them.

VladVidya, 2017-09-07:
Sorry for the late responses. I don't use Github often.

Done, and no success.

ShadixAced, 2017-09-07:
#888 's got updated recently due to some feedbacks. Give it another shot !

closed 2018-01-26 by nefarius