nefarius/ScpToolkit #894: Controller being stupid. pls hlp.

Ritikkhilnani, 2017-08-22:
When I plug in my controller the lights blink but nothing happens. I tried everything on YouTube and nothing works. A few months ago I tried it and got frustrated so I formatted my computer. I forgot about it until now. I tried to do everything calmly now and luck isn't on my side. Help would be greatly appreciated.

Ritikkhilnani, 2017-08-22:
When I plug in my controller the lights blink but nothing happens. I tried everything on YouTube and nothing works. A few months ago I tried it and got frustrated so I formatted my computer. I forgot about it until now. I tried to do everything calmly now and luck isn't on my side. Help would be greatly appreciated.

ShadixAced, 2017-08-22:
Which version of Windows are you using ?

closed 2018-01-26 by nefarius