nefarius/ScpToolkit #897: How do I forget/remove/uninstall a controller??

thefowles1, 2017-08-24:
I have been using ScpToolkit with a DS3 controller for a while. A friend of mine wanted to get their PC set up to also use a DS3 controller with ScpToolkit. So when I got a second controller for them to use, I tested it with my PC, and now the controller always connects to my PC as controller #2, even when their computer is in range. No matter what I try to do to get my PC to forget this controller, it always reconnects. I have attempted to delete the Playstation 3 controller driver but it always reconnects after rebooting. Do I need to run the clean wipe utility or is there some other way to get my computer to just forget this second controller???

thefowles1, 2017-08-24:
Update: decided to say screw it, used the clean wipe utility, then downloaded the latest 1.7 beta and used that. But now, BOTH controllers connect via bluetooth as controller #1 at the same time...

thefowles1, 2017-08-24:
I'm really frustrated about this and wish I hadn't tested the controller on my PC because I can't get my PC to just forget this controller. That's all I want to do. Just have it work with my controller only. I tried starting fresh with the clean wipe utility but it's like this second controller is now permanently attached to my PC...

SneakySnake04, 2017-12-22:
Bump. Someone, help.

closed 2018-01-26 by nefarius