nefarius/ScpToolkit #911: Bus Open: Offset 0 Failed / Game doesn't recognize my ps3 dualshock

Franklinitro, 2017-09-06:
I use Scp in a while and I'm used to fix the errors myself, but yesterday I came up with an error I didnt found any solution. When I started the server, the program showed this message: "Bus Open: Offset 0 Failed", I thought it was a simple error, because my dualshock connected with the program, but when I open a game, it didn't recognise my genuine ps3 dualshock. I even open the server to see what's wrong, but besides this "Bus Open: Offset 0 Failed" everything seems working.

[image]

Franklinitro, 2017-09-11:
I already found a solucion.
The problem was on the scp bus driver, it was with a yellow mark, you can find it in device manager. So I disable security boot on the BIOS, and then I just open the scp setup and clicked on repair. Now everything is OK.

Flimzee, 2017-09-18:
I think I'm having the same issue. Everything seems to working but when I run pes 17 the controller doesn't work. There is no yellow make on my scp bus driver.. I'm confused pls help

kamer1337, 2017-11-15:
installing 1.6 version resolves the problem

augustonfs, 2017-11-19:
I'm having the same problem, --Bus open: offset 0 failed.
The thing is that I have version 1.6, and was using it just fine a few days ago.

Ultroman, 2017-12-19:
Same as augustonfs, except I've been running 1.7 for months...one day, out of the blue, "Bus Open: Offset 0 Failed", both my controllers show themselves to be controller 1 (the red lights on them), but in SCP Monitor they show as Pad 1 and Pad 2, both charging happily. But none of them work anywhere.

closed 2018-01-26 by nefarius