nefarius/ScpToolkit #915: Non-blocking bluetooth and update service

Gibtnix, 2017-09-10:
Hello!

First of all, thanks for the great work!

However, I recently switched from an old SCP toolkit version (which I used for ages to use my DS3 controller) I downloaded somewhere to the 1.6.238.16010 release version because it includes support for my notebook's internal bluetooth device, for the old SCP toolkit I had to use an USB dongle since the internal one has not been recognized. Everythings works like a charme, especially the installation process is much simplified in comparsion with the previous version, great work!

But even though my internal bluetooth works now, the scp toolkit hijacks my whole bluetooth adapter and replaces it with a libusb device and I cannot add further bluetooth devices, Windows simply does not recognize that my computer has a bluetooth adapter. I tried to find some information about this, but from what I've found out I conclude that this seems to be the expected behavior, i. e. I won't be able to use other bluetooth devices any more after I installed the SCP toolkit? Can or will this be fixed sometime, i.e. it simply uses Window's bluetooth device without replacing the driver with a libusb device?

Second I would like to ask for an option to disable the auto update... I can disable the service autostart, but this will also include that my controllers aren't automatically detected when I try to connect them via the PS-button, right? Hence, I would like to keep the service but disable the auto update, first for performace reason (unwanted delay from internet connection) and second the popups are simply annyoing ;-)

Appreciating any feedback and once again, great work!

Zortothan, 2017-10-05:
For your bluetooth problem, this should help: https://github.com/nefarius/ScpToolkit/issues/599 
It fixed mine.

Gibtnix, 2017-10-05:
Thanks for your comment! I followed the respective instructions, but after that the controller does not connect itself via bluetooth... I think this solves occupied wifi cards, but could you remove the respective bluetooth device and still connect your controller via bluetooth? For me this seems not to work at all... :-/

closed 2018-01-26 by nefarius